AguaClara/ram_pump #1: ram pump and spring forces

monroews, 2018-02-06:
It is possible that the forces required will work if drag on the valve while it is opened is reduced by opening the valve much further. Then you can use a relatively long and weak spring such that the force is reduced in both open and closed positions.

I suggest setting up the ram pump and testing it as soon as possible to see if you can get different cycle responses including the response of staying closed for a fraction of a second.

mailemccann, 2018-02-07:
We tried moving the top stand off down to increase the amplitude of the plate movement and keep the valve open longer, but the plate is too short to move further out, as the threaded rod has too large of a diameter to fit through the valve. We were able to move the plate out 1cm. We either need a longer plate or a thinner rod to effectively integrate a longer spring

monroews, 2018-02-07:
1 cm sounds like plenty of travel. Try that. 
The goal isn't to keep the valve open longer. The goal is to keep the valve closed longer. It is when the valve is closed that the high pressure pumping happens.

mailemccann, 2018-02-07:
I meant to write plate travel further*

closed 2018-04-20 by mailemccann