AguaClara/ram_pump #10: Brainstorm ways to change the RAM Pump to increase efficiency

stevenlopez182, 2018-02-21:


mailemccann, 2018-02-23:
Current ideas to move forward with:
-Horizontal configuration with swing check valves
-Addition of magnets
-Double sided hook spring
-Inverted configuration

mailemccann, 2018-02-24:
Why are we not concerned with efficiency? Our current configuration that we have been running cycles normally, and effluent is being pumped but the ratio of water being pumped to the water being wasted is very low. We are using a spring with a lower k value than had been used in past semesters, and from comparing the two videos the lower k value is staying closed longer but still is inefficient. 

I also thought we decided that our current configuration with one spring is inefficient and we likely need to significantly change the configuration. I see it as two separate ways to proceed, or both simultaneously:
1. Change the configuration 
2. Work to perfect our mathematical modeling to determine right spring force

monroews, 2018-02-24:
I hadn't realized that you had succeeded in getting the valve to stay closed longer. How long is the valve staying closed? How does that compare with the theoretical deceleration time?

If you have solved the problem of the valve not staying closed, then we may not need to make significant changes to the pump design.

Then we can think about what might be causing inefficiency. One possibility is that the pipe sizing for the high pressure check valve might be too small. When the large waste valve closes the initial flow rate through the high pressure check valve is quite high and there is excessive head loss. This results in the column of water decelerating much faster than we had hoped.  Thus a solution might be to increase the size of the piping on the high pressure side up to the entrance into the air chamber.

monroews, 2018-02-24:
It might be good to revisit our definition of efficiency. You can calculate how much water can theoretically be pumped in a cycle given the flow rate at the moment the waste valve closes. Compare the flow that you create with that theoretical value to get a better measure of efficiency.

mailemccann, 2018-02-24:
We didn't measure time closed we measured effluent/waste water, which was only about 6.6%. We didn't make too many changes to the old design yet though, just used a weaker spring and changed the connections between the two check valves, but the previous value for effluent/waste water was about 5% I believe, which isn't that significant a difference. I think you can hear the difference on the videos though. We can measure the time closed this week

I can't seem to upload videos here but I will email them to you!

mailemccann, 2018-02-24:
So efficiency may be better measured as 
actual/theoretical effluent flow
vs
effluent/wasted water?

monroews, 2018-02-24:
Ram pump efficiency is never measured as effluent/wasted. I believe it is common to measure it as (Q_pumped * H_pumped)/(Q_wasted * H_drivepipe). This way efficiency is a measure of energy. I'm curious if the method I proposed to measure the amount that can theoretically be pumped in one cycle is similar or different from the energy efficiency.

stevenlopez182, 2018-02-27:
[image]

closed 2018-04-20 by mailemccann