AguaClara/ram_pump #21: Equation for Plate Amplitude and Drag?

mailemccann, 2018-03-14:
@monroews 
Do you have an equation that related amplitude to drag force? I know only of the drag equation over a flat plate using velocity and reynolds number... and we are slightly stumped on how increasing the amplitude effects this equation

stevenlopez182, 2018-03-14:
Would it be possible to add a spherical surface on top of the flat plate to reduce the drag? Also if this is viable, would adding dimples to this semi spherical surface reduce the drag even further?

monroews, 2018-03-14:
Our goal is to have longer time with the valve closed. This means that we need to have LESS spring force pushing up on the valve when it is closed. That means that we need to have a weaker spring. Thus in order to keep the valve OPEN for a reasonable time we need to have a small spring force be able to keep the valve open. Thus we need low drag. The drag on a plate is set by the low pressure in the wake. The low pressure in the wake is set by the velocity at the point of separation. The velocity at the point of separation is high when the valve is only opened a little bit and is lower when The valve is opened further. Thus a weaker spring (this is our goal) is possible if the valve is opened further. I'd recommend using a long weak spring to hold the valve open far and see how it works. If you get to the place where the valve doesn't  open again, then you have gone to far and you need a stronger spring. If the valve doesn't stay closed long enough even with the weak spring, then you can explore reducing drag on the valve with the sphere.

closed 2018-04-20 by mailemccann