AguaClara/ram_pump #23: 3/20 Force Calculations

mailemccann, 2018-03-20:
- Using

 python code, calculated hydrostatic force when plate closed to be 2.5 N

- Max

 drag force when plate is completely open is 10.56 N as per last semester's Python Code

- Need

 to ensure that plate can open completely- must overcome hydrodynamic force while plate is partially open

Calculations below:

[image]
[image]

stevenlopez182, 2018-03-20:
I agree with this

mailemccann, 2018-03-20:
Thanks for the support @stevenlopez182

mailemccann, 2018-03-20:
Goals for next meeting:

- Measure max displacement (current amplitude)

- Decide k value that can resist 2.5 N but is closed when 10.56 N applied

- Order new springs in range

mailemccann, 2018-03-21:
3/21 - All goals completed

closed 2018-04-20 by mailemccann