AguaClara/ram_pump #28: Efficiency Testing 4/11

mailemccann, 2018-04-11:
k value = .46
length = .062 m

Waste volume: 2.6 - .75 = 1.85 L
Effluent Volume: .76 L

mailemccann, 2018-04-11:
Fall 2016 Research Report, where we extracted values for effluent head

https://confluence.cornell.edu/display/AGUACLARA/Ram+Pump?preview=/152026512/340901334/RamPumpFinalReportFall2016.pdf

monroews, 2018-04-13:
This efficiency are incredibly high. Was there a consistent 10 m of head on the pumped water side?

mailemccann, 2018-04-13:
No, the 10 m head is an averaged value and it was actually extracted from the Fall 2016 research report in their efficiency testing. We had very inaccurate readings for effluent pressure due to excessive noise in the pressure sensor reading and low sensitivity on the 200 and 7kpa pressure sensors. 

The efficiency values were alarming to us and we are meeting today to go through more iterations of the efficiency test and try and obtain a more accurate pressure reading for the effluent

closed 2018-09-30 by AlyciaStorch