AguaClara/ram_pump #35: Update on Python Code

mailemccann, 2018-05-04:
The past week we have been working on a Python code to find the idealized maximum efficiency for the pump under a set of experimental conditions. This allows us to see how far off we are from that maximum efficiency and what the biggest contributors to loss of efficiency are.

The goal of the rest of the semester is to set up our code and calculations so that next semester's team can work more efficiently and have a faster learning curve.

mailemccann, 2018-05-04:
We are also working on analyzing experimental pressure data via ProCoDa_parsing python code in AguaClara_research repository in order to determine our efficiency.

WillLopez, 2018-05-04:
Additionally, we measured effluent head via ProCoDa and will be including the data analysis in our code

closed 2018-09-30 by AlyciaStorch