AguaClara/ram_pump #46: Set up Ram Pump Pulley System

cheertsang, 2019-02-07:
- [ ] Add sliding platform for balance
- [ ] Figure out how to get sump pump to start pumping water to head tank
- [ ] Test out weights

cheertsang, 2019-02-10:
@monroews we are a little confused on the point at which we measure the mass reading as we lower balance because there isn't really a clear range where the mass reading is constant. Our understanding is that we are using the balance to measure the tension force of the string, so initially the string is exerting no tension force on the bottle. As we lower the balance, the tension force is increasing. At some point during this, the valve opens, which would presumably relieve the tension force briefly?

monroews, 2019-02-10:
Presumably as soon as the valve stem lifts off of the valve seat the balance is finally reading the weight of the  valve stem. Then if you lower the balance slightly you should get the SAME reading because the valve seat is still suspended by the balance. But if you lower the balance too far, then the valve seat will hit the stop that prevents it from going any higher. At that point the reading on the balance will drop.

monroews, 2019-02-10:
Opening the valve doesn't relieve the force! once the valve begins to open you should see the maximum force.

ChingPangggg, 2019-02-11:
We connected the pulley system to the balance using a bottle filled with water as our mass. However, we find it difficult to find the force required to lift the valve still. When we were lowering the the balance, we found it difficult to precisely adjust the height even with the use of a handle, because moving the height very slightly changes the reading by a lot. We also cannot find the period where the reading remains constant. Would you please advise us on this? @monroews

ChingPangggg, 2019-02-11:
We found the approximate point to be 136g but we are not sure about this. In addition, last semester, the team proposed another method to measure the force 

> The bottle had enough water in it to keep the plate open while the pump was running. Then, with the pump still running, the team emptied the bottle little by little until the plate was no longer held open by the bottle. The team then used a pipette to slowly add water back into the bottle until the plate was held open again.

Do you think that this method would be more accurate?

monroews, 2019-02-12:
Now that you have the balance at the correct height so that the balance is supporting the weight of the valve stem, now you can vary the flow rate and measure the change in force on the valve stem. Now that the balance is at the correct height it should be possible to get all of the measurements quickly.

monroews, 2019-02-12:
You can check the weight by measuring the mass of the valve stem in the air and then correcting that for buoyancy. You can find the density of stainless steel online.

closed 2019-03-16 by cheertsang