AguaClara/ram_pump #47: Run Experiments with Pulley System

cheertsang, 2019-02-07:
- The spring is removed.
- The force of moving water will initially be exerting a downward force on the plate.
- We will slowly add water into the bottle tied to the pulley until the weight of the bottle overcomes the downward force of moving water, opening the plate and allowing water to flow out of effluent valve.
- At this point, we will measure the weight of the bottle. Knowing the weight of the rod and plate, we can calculate the force of moving water: Fwater = Wbottle - Wrod. 
- Knowing Fwater, we can calculate the spring constant by setting Fwater = kx, where x is the distance of compression. 
- At the same time, we will collect data on the headloss at the effluent valve to measure the efficiency of the system, where higher headloss indicates higher efficiency, as it allows water to travel a greater height. 

- [ ] Measure weight
- [ ] Measure headloss with pressure sensor

cheertsang, 2019-02-12:
Both buckets on the floor should be almost full when head tank above is empty. 
When water in head tank reaches line marked, throttle valve pumping water up to head tank. This should keep the water level in the sump pump bucket so that the intake is submerged. Overflow pipe from head tank should be draining slowly, keeping water in head tank at a constant level.

cheertsang, 2019-02-13:
Add relay box to turn on/off sump pump, control with ProCoDA. (Like setting up mixer)

ChingPangggg, 2019-02-21:
Run experiments after replacing string with wire. 
Opened all valves. Close valve at bottom of ram pump. Start sump pump. Throttle valve from sump pump to head tank. When water level reaches line in head tank, turn off sump pump. 

Changed out scale. 

98.6 g, 87.4 g, 82.4 g, 81.4 g

ChingPangggg, 2019-02-28:
2/28/2019
84.7 g, 102.5 g
84.0 g, 86.5 g
Measurements have been inconsistent and we are suspecting that this is because how the wire is bendable

ChingPangggg, 2019-03-04:
displacement of plate; 
closed from top of threads: 3.5 cm
open: 3.0 cm

mass of plate and hook: 24.9 g

closed 2019-03-16 by cheertsang