AguaClara/ram_pump #50: Buy new steel cable for pulley system

cheertsang, 2019-02-12:
Find cable < 1.74 mm (0.06850394 in)

**Cable Galvanized Steel Coated**
Cable Material Galvanized Steel Coated Coating Material Nylon Strand Type 7 x 7 Cable Size 3/64 In. Outside Dia 1/16 In. Length 50 ft. Working Load Limit 54 lb. Standards MIL-DTL-83420 TYPE 1 Includes Zip Tied Coil Manufacturers Warranty Length Lifetime Wire Rope End Type None

[eShop](https://solutions.sciquest.com/apps/Router/GeneralCatalogItemDetailsPopup?ItemId=295283662&ItemSource=0&SearchInputs=%7B%22Keywords%22:%22steel%20cable%22%7D&RowIndex=28&SearchTransactionId=1550012020495449&SourceContext=SearchResults&formRequestShoppingEntityType=&Quantity=1&tmstmp=1550012168686)

cheertsang, 2019-02-12:
try consolidated electronic wire (26 BNC): 0.41 mm diameter

If this doesn't work, look for flexible cable around this diameter.

ChingPangggg, 2019-02-28:
We tried using fishing wire and various ropes, but they were all too elastic. (The fishing wire was even more elastic than the string). The 26 BNC consolidated electronic wire was too bendy, which made measurements inconsistent because it held its shape. We think it is important that the material we use doesn't stretch by more than a few millimeters because the difference between opening and closing the valve is only a few millimeters. Any suggestions on what other materials we could try? @monroews @mailemccann

monroews, 2019-03-01:
I played with your system a bit yesterday by gently pulling on the wire and I could feel a initial resistance that prevented the valve from lifting. I'm wondering if there is something that is causing the valve to not slide easily. Is the force lifting the valve not aligned with the center of the valve?

I think the wire is springing because it is too stiff. Thus a much thinner wire would likely be better. You can easily check the elasticity of steel to see if this is correct.

What prevents you from giving the valve more like a centimeter in travel? Would that make it easier?

I'm worried that we are more than one month into the semester in what was supposed to be about a week's effort and we aren't getting closer to our goal. I suggest it is time to reassess strategy and figure out how to make progress on multiple fronts. There are multiple motivating factors. In order for your team to win an EPA phase II award you need to have progress and great ideas for the proposal. We also want the ram pump to be a reliable component in our water treatment plants. Perhaps hold a team brainstorming session to explore how you could get to those goals. 

Also, make sure you have the supplies on hand to manufacture at least one more ram pump so that if there are any problems with the first pump that you can keep on moving ahead.

closed 2019-03-09 by ChingPangggg