AguaClara/ram_pump #52: RA Comments on Fab Manual 1

mailemccann, 2019-02-27:
Good stuff guys. Lets meet next week to discuss my comments if possible. 

Abstract-
You mention the "valve" without explaining which valve, what the valve is, etc. I think you can either broaden this statement to say "optimize the cycling times" or something slightly less specific, or explain what valve you're talking about. 

Introduction-
Here would be a good time to introduce the concept of "head" instead of using the term "falling water". This is more descriptive of the physical process that is being utilized. I also think this section could be better organized- propose a problem and explain how the ram pump is solving it, instead of jumping around. Also chemical stock tanks can ideally be explained more clearly before referencing them.

Expand on "inline" to explain that its vertical and can be inserted into an already existing pipe system. This is important to emphasize since its what sets our design apart.

Lit Review-
If you're going to discuss a conventional ram pump (which I think was a good idea!), explain the concept and structure a little more in depth, and why its different from AguaClara's. Also you mention the air chamber in the figure description, but it hasn't been explained. Write a description of what the air chamber is used for here (to maintain a pressure constant and allow for continuous flow). The next paragraph should be in previous work. Maybe include your review of Monroe's power point on hydraulic transients, your review of some of the underlying concepts like head loss and driving head, etc. 

Previous Work-
Again, explain what the check valve is before you reference it. You could include way more info about what last semester did. Also, the first paragraph is not entirely accurate. The model does not map forces in order to design a toggle mechanism, it models the existing spring system and looks for ideal velocity and maximum efficiency of the system AguaClara has been using. If you want to discuss the idea of using other toggling mechanisms, maybe do that in the introduction or body, but be sure to emphasize that the mechanism that we have been working on optimizing is a spring. 

Lab Setup-
Your language is slightly imprecise. Try to use the same terminology throughout- eg: drive pipe not drive pump. Falling water is not very descriptive of the actual process. Also, the pressure sensor is not measuring "headloss" its measuring pressure that the ram pump is pumping to. This pressure can then be converted to an elevation head, not "headloss", to see how high the effluent can be pumped to. I think for the next report, it would be good to rewrite this section to follow the flow of water a little more clearly, so that the reader can get an idea of the steps of the pumping process and what each mechanism in the set up does. 

ALSO add a section about the internal setup inside the valves! This is super important in explaining how the ram pump operates, and is important that the reader understands this before looking at the force balance diagram in Figure 6. 

Water Velocity Section-
Explain why maximizing the area under the curve is beneficial. Explaining Monroe's hydraulic transients power point in lit review might be helpful to then reference here. 

Forces Section- 
Nice! Good technical writing and easy to follow. Try to explain some of this is your previous work as well. I'm not a huge fan of the aesthetics of the force balance figure, with the giant purple arrows...Maybe clean this up a bit before the next report. 

Fab Details- 
This Onshape drawing is great, maybe show how this fits into the valves, however, and explain the internal setup a bit more. You guys have the necessary information to expand on this section also, so I think you could write more here.

sidneylok, 2019-03-07:
Hey guys! Just wanted to add that you should have a bibliography/references section to compile all of citations for the reports and papers you referenced in your lit review and previous work sections. I didn't grade this in your first manual because I'm not sure if it was made clear enough that Fab teams need to include this, but you will be graded on it next time so be sure to include it in your next iteration! Thanks :)

cheertsang, 2019-03-14:
Revisions for Draft 2:
- [ ] @alyssaju **Revise Lit Review:** add descriptions of conventional ram pump and how the inline ram pump differs (conventional ram pump is described in EPA grant [Project Description](https://docs.google.com/document/d/1g4uX_CjjWllurb4KftwjLmaFy6Au-42kGi5e5m47YEI/edit#heading=h.o37szg9orcu4)), address comments
- [ ] @inigocabrera44 **Revise Previous Work & Lab Setup:** address comments
- [ ] @ChingPangggg keep working on OnShape diagram 
- [ ] @cheertsang **Work on Methods Section:** material and tool list, procedure steps/experiment steps
- [ ] Experiments/Results: pressure graphs, calculations
- [ ] Future Work
- [ ] Conclusions

@sidneylok which components of the [research report template](https://github.com/AguaClara/team_resources/blob/master/Report_Template.md) should be included in the Manual template?

sidneylok, 2019-03-15:
Wow this checklist is so cool! 

You should at the very least have the Abstract and Introduction, but Previous Work and Lit Review should be included as well. Seems like you're already doing that. The other components of the research report can be added in as you see fit. For instance, StaRS FInE is including the Methods section because they ran some experiments and want to explain them there and give Results and Analysis. Because they are using the Methods section, they probably won't need to include Experimental Methods as it would be repetitive. Though, they are a bit special because they are more research-y than Fab. 

tldr; You can add stuff from the research report template as you see fit. 

You should have all sections completed or at least started by this point. Draft 3 will build on these sections by adding on new work or findings. If you are unable to deliver that, just let graders know.

closed 2019-03-16 by cheertsang