AguaClara/ram_pump #57: Goals

ChingPangggg, 2019-03-05:
Weekly:
Week of 3/4/2019:
Ching: Order parts to fabricate a new ram pump

End-of-Semester Goals:

cheertsang, 2019-03-05:
Weekly Goals: 
- [ ] Edit Research Report
- [ ] Understand efficiency graph
- [ ] Run tests by pulling on valve manually, analyze pressure graphs

Standard pipe sizes for each plant?

alyssaju, 2019-03-05:
Weekly Goals:

- [ ] Learn/research more about fluid dynamics and physics of Ram Pump
- [ ] Understand code of mathematical model (efficiency graph)

inigocabrera44, 2019-03-07:
Week of 3/4/2019

- [ ] Learn/Understand the pressures regenerated at the Pump (including air chamber).
- [ ] Run the test and learn how to set up the rum pump with ProCoda and how to transfer and collect the data to Atom at our laptops.
- [ ] Empirical calculation of forces

closed 2019-06-03 by ChingPangggg