AguaClara/ram_pump #58: Pressure cycles observed with setting initial pressure

ChingPangggg, 2019-03-09:
We set the set the air chamber's initial pressure to be 600 kPa, and we ran the system by manually control the opening and closing of the valve. We obtained the graphs as shown below:
[image]
[image]

ChingPangggg, 2019-03-09:
@monroews Could you please advise us on what should we interpret from it? I am also a bit confused  about what we should be looking at from the wasteline pressure graph. Thank you very much.

monroews, 2019-03-09:
Share the graph in the issue so we can discuss it

cheertsang, 2019-03-11:
@ChingPangggg the pressure was 600 cm, since I added both pressure sensor with units in cm of water (Both are 200 kPa sensors).

ChingPangggg, 2019-03-11:
@monroews We actually pressurized the air chamber to be 600 cm, and then start running the experiment. Would you mind clarifying how pressurizing it initially can help us observe the cycle? Are we pressurizing it because we want to to allow the water to travel up to 6m (600cm), and we want to observe the pressure change if the air chamber starts at 600 cm of pressure?

We are calculating the water being pumped using ideal gas law with delta P being the change of pressure in the air chamber at each cycle, and then using delta V=nRT/dealta P, we can find delta V as the change of volume of the air in the chamber, and then how can we relate the initial volume of air in the chamber to find the volume of the water bring pumped into the chamber in each cycle? For our experimental setup, how should the change in volume of air and change in volume of water?

monroews, 2019-03-11:
reason to pressurize to 6 m is to mimic what the ram pump will be doing in a real plant.
You want to know how much water the ram pump can pump when lifting water 6 or more meters.

change in air volume = - change in water volume
Set up this analysis in python. Show me what you find!

Also don't forget to look at the pump part of the cycle in more detail to see the time when the water is decelerating in the drive pipe.

mailemccann, 2019-03-11:
@ChingPangggg I don't think you need to relate to the initial volume in the chamber; you're just looking for the delta V for one pump cycle, eg: how much water was pumped in that individual cycle

monroews, 2019-03-11:
You will need to know how many moles of air are in the chamber. You get that by knowing the pressure and volume at some point. Can be initial or final or random. Happily mass is conserved.

cheertsang, 2019-03-16:
[image]

We took one cycle and found the pressure change in the air chamber using this code:

`import aguaclara.core.physchem as pc
import aguaclara.core.constants as c

#find volume of air using ideal gas law
#deltaV = nRT/deltaP

#pressure difference, deltaP (measured as difference in pressure in air chamber)
url = 'https://raw.githubusercontent.com/AguaClara/ram_pump/master/Spring%202019/3-7-19_manual_operationwith650Pitrial2.xls'

start = 7000 #should be more than 'start'
end = 10000 #should be less than 'stop'

airchamber = pp.column_of_data(url, start, 2, end, 'cm')

head = airchamber[-1]-airchamber[0]

#n, initial number of moles of air
diam = 1*u.inch
height_water = 1.5 * u.inch #initial height of water in air chamber
vol_water = (pc.area_circle(diam)*height_water).to(u.m**3) #initial volume of water in air chamber
height_ac = 26.5*u.inch #height of air chamber
vol_airchamber = (pc.area_circle(diam)*height_ac).to(u.m**3)

init_vol = vol_airchamber - vol_water #initial volume of air in air chamber
mass = (c.AIR_DENSITY*init_vol).to(u.kg)
mol_mass = 28.97*u.g/u.mol

n = (mass/mol_mass).to(u.mol)

#gas constant
R = 8.314 * u.J/(u.mol*u.K)

#standard temp
T = 293*u.K

#convert head to pressure
deltaP = pc.density_water(T)*c.GRAVITY*head

#calculate change in volume of air, which is equal to the volume of water pumped
deltaV = (n*R*T)/(deltaP).to(u.Pa)`

I'm having trouble with the units because `deltaP` is in cm in `deltaV = (n*R*T)/(deltaP)` 
I tried converting the head in cm to units of pressure in Pa using `deltaP = pc.density_water(T)*c.GRAVITY*head`

cheertsang, 2019-03-16:
@monroews We figured out that the [constants.py](https://github.com/AguaClara/aguaclara/blob/master/aguaclara/core/constants.py) package provides the 1.204 kilogram/millimeter^3 as the AIR_DENSITY, when this should be 1.204 kilogram/meter^3.

monroews, 2019-03-16:
Convert to units that are easy to understand. mL?
Post an issue on aguaclara repo re the error in air density. Propose that air density should NOT be a constant! It should be calculated from PV=nRT.

Add units to your x axis!

Create a graph that shows just the pump cycle. The pump cycle likely lasts a fraction of a second and corresponds with the increase in pressure in the air chamber. See if you have enough data to describe that event well. You might need a higher data rate. I'm a little confused about the pressure trace for the drive pipe. Is it possible that you are plotting the negative of the pressure?

cheertsang, 2019-03-19:
@monroews  I updated the graphs above so that the pressure is positive for both sensors and added units to the x axis. This makes sense to me for the air chamber pressure, because when the valve closes, the pressure in the chamber increases as water is pumped in.

However, I am confused as to why the pressure in the waste valve has a positive spike when the valve is closed. **What insight can we gain from the waste valve pressure traces?**

monroews, 2019-03-19:
The pressure traces suggest that the pressure above the waste valve was never high enough to pump any water into the air chamber. I'd check how the sensors were zeroed and suggest checking to see if they read the same value when measuring the same pressure.

I'd propose that mL is easier to understand than cubic meters for the flows you are generating.

Once you have that corrected pressure trace then it should be possible to see the length of the time that the waste valve pressure exceeded the air chamber pressure. You should be able to compare that duration with the length of the pump cycle that you predicted based on F=ma. You can also compare the volume pumped with the predicted volume of water pumped based on F=ma and Q=VA.

cheertsang, 2019-03-21:
We zeroed the pressure sensors at atmospheric pressure. We also checked to see if they read about the same value when measuring the same pressure, and they do. We also made sure the pressure sensors were attached the right way so that we didn't get negative pressure readings. 
[image]

**From my understanding, the waste valve pressure should reach the air chamber pressure in order to pump water into the air chamber? Is it not reaching the air chamber pressure because we pressurized the air chamber at 600 m and there isn't enough head to pump water into the air chamber? **

monroews, 2019-03-21:
So now run some pump cycles again and track the air chamber pressure to see if you are pumping water.

cheertsang, 2019-03-21:
Water is definitely being pumped into the air chamber, so the waste valve pressure should reach the air chamber pressure, but it is not doing that

monroews, 2019-03-21:
Are you getting pressure traces that show when the valve cycles?

cheertsang, 2019-03-23:
Are the graphs we generated above the pressure traces when the valve cycles?
And the one below is showing the pressure of one cycle?
[image]

monroews, 2019-03-23:
That graph looks like old data from before you zeroed the pressure sensors.

cheertsang, 2019-03-23:
We zeroed the pressure sensors before water started running through the system. This graph below shows the pressure when water is running through the system. Should we zero it at this point?
[image]

cheertsang, 2019-03-23:
We zeroed it when the water started running through the system, but the waste valve pressure still never reaches the air chamber pressure, even though we can see that water is being pumped into the air chamber.
[image]

monroews, 2019-03-23:
You know that this pressure data can't be right.
Hypotheses???

- one or more of the pressure sensors isn't calibrated correctly. This would mean that if both pressure sensors were connected to the air chamber they would report different pressures.
- The high speed pressure pulse doesn't reach the waste valve pressure sensor because it is attenuated by the flow of water and compression of air in that sample line. If this is happening (you can watch the water moving in this tube while the ram pump is cycling!) then you could eliminate this problem by filling this tube the whole way (minus the last 5 cm or so).

ChingPangggg, 2019-03-23:
We've checked the calibration of the pressure sensors by connecting them to the air chamber and they read the same pressure, so we are fairly certain that this is not the issue. 

**Should we zero the pressure sensors before we start running water through the system? Or once we've started running the system and filled up the air chamber almost all the way?**

We have observed the water in the air chamber increase every time the valve closes, so we know for sure that water is being pumped into the air chamber. We will try running the system by filling up the air chamber almost all the way, and see what we get.

monroews, 2019-03-23:
The pressure sensors should be zeroed when they are both at atmospheric pressure.
Please check my second hypothesis. The high speed pressure pulse doesn't reach the waste valve pressure sensor because it is attenuated by the flow of water and compression of air in that sample line. If this is happening (you can watch the water moving in this tube while the ram pump is cycling!) then you could eliminate this problem by filling this tube the whole way (minus the last 5 cm or so).

ChingPangggg, 2019-03-23:
We zeroed both sensors a atmospheric pressure, and then filled the air chamber all the way, leaving the last 8 cm. However, the waste valve pressure is still much lower than the air chamber pressure.

[image]

ChingPangggg, 2019-03-23:
We also did another trial, and we obtained the total volume of water pumped for the duration of our data (10 cycles) to be 0.099 L, which is around 10mL per cycle.
[image]

monroews, 2019-03-23:
Please check my 2nd hypothesis

ChingPangggg, 2019-03-26:
We shortened the tubing connecting the pressure sensor at the waste valve position. We ran a test again and obtained the following graphs:

We manually opened and closed the plate for 16 times (timing is selected arbitrarily), and we found that approximately 0.2149 L was pumped in total, which means that approximately 0.0134 L was pumped per each cycle.
[image]

We ran another trial, and the total volume pumped is 0.5182L, and the volume pumped per each cycle is 0.0324 L.
[image]

monroews, 2019-03-26:
Excellent!
Can you show full detail of one pump cycle? You may need to increase the data rate (both acquisition rate and rate of saving to file).

How much air do you have in the pressure sensor line?

closed 2019-06-03 by ChingPangggg