AguaClara/ram_pump #64: RA Comments Fab Manual #2

mailemccann, 2019-03-28:
Hey!! AMAZING additions to your report! Here are my comments:

- Nice job with the abstract and introduction! I think you explain the purpose of the ram pump really well
- Lit Review: is "header tank" in the image description supposed to be "head tank"? Idk if a conventional ram pump uses different terminology or this is a typo
-- I LOVE the lit review section. Very comprehensive, and makes it really evident why the vertical ram pump is so important in AguaClara plants vs conventional technology
-- One thing you could emphasize is that the plate DOES NOT drive water, its the pressure difference that pushes water out the effluent; I only mention this because this is one of the biggest misconceptions with ram pump and it might be worth mentioning for other people reading your report with less background knowledge

-  Ok I read the rest of the report and I am honestly blown away. You guys did a GREAT job. I'm going to read again and look for nitpicky stuff and reply to this issue. Structurally and content wise I think this is perfect.

mailemccann, 2019-03-28:
- Under fabrication details, it would be helpful to have where each component can be purchased, for future teams to reference
- Procedure: keep your sentence structure consistent from the introductory paragraph to the list: eg: "Pressurize the pump to __ prior to..."
- Oh, under experimental methods, you have written (insert picture) but then forgot to insert the picture! This would be a good section to have as many visuals as possible
- Pressure Cycles Analysis: the change in air volume is the negative change in water volume
- I'm looking at the rest of the report with bias, as we figured out why the check valve pressure was less than the air chamber pressure in those graphs. From the data you had at that point, this is good. We can work on how to write into your final report the error that you made and how it skewed the data

cheertsang, 2019-04-11:
- [ ] @ChingPangggg address comments in fabrication
- [ ] write section about finding experimental terminal velocity in methods
- [ ] results of terminal velocity experiments
- [ ] fix graphs and code in report
- [ ] write code for calculating theoretical amount of water 
- [ ] efficiency calculations

closed 2019-05-07 by ChingPangggg