AguaClara/ram_pump #65: Things to Do

cheertsang, 2019-03-28:
- [ ] Verify terminal velocity by experimentation
- [ ] Calculate theoretical water pumped
- [ ] fix code for actual calculations
- [ ] Calculations to find efficiency (actual/theoretical)

experiment with different parameters of experimentation (time of opening/closing valve)
look at graphs at point where waste valve pressure goes above the air chamber pressure

cheertsang, 2019-04-12:
Height of drive pipe: 1.74 m
Height of drive pipe + water level in head tank: 2.06 m
diameter of drive pipe: 23.65 mm

cheertsang, 2019-04-16:
- [ ] adjust pressure sensor - move closer to effluent
- [ ] re-run tests, measure height of water in air chamber - get a rough estimate of how much water is being pumped during each cycle

cheertsang, 2019-04-16:
initial water level 6 cm from top of air chamber
initial pressure in air chamber: 698.97 cm
final water level 5 cm from top of air chamber

alyssaju, 2019-04-18:
**Preparations for EPA Conference (RamPump)**

- [ ] Make a poster to present
- [ ] Interactive activity showcasing rampump (check if station has electricity)

cheertsang, 2019-04-20:
**Meeting with Monroe**
buy new pressure sensors, one that allows for higher pressures
redo pulley system tests to figure out what spring we need
- set the length of spring, solve for k
build new ram pump - effluent same size as waste check valve
- need to order brass nipple?
- something about the connection btwn the check valves

figure out setup for EPA

alyssaju, 2019-04-23:
4/25 Meeting
- [ ] Write EPA grant!!!

alyssaju, 2019-06-04:
**Week of 6/3**
EPA Grant Proposal:
- [ ] Send out emails for letters of support
- [ ] Finish Phase I of grant proposal
- [ ] Brainstorm for Phase II of grant proposal
- [ ] Complete and compile forms for grant proposal

Lab:
- [ ] Cut tube for air tank and glue onto effluent pipe 
- [ ] Secure top third of ram pump to the drive pipe and the air tank to the frame

**Week of 6/10**
EPA Grant Proposal:
- [ ] Begin writing Phase II and continue brainstorming project ideas for Phase II of grant proposal
- [ ] Compile letters of support on Google Drive

EPA Conference:
- [ ] Work on poster (link on team Google Drive)
- [ ] Create checklist of what to bring to EPA Conference (e.g. parts of ram pump)

Lab:
- [ ] Install new pressure sensor 
- [ ] Run water through new ram pump; test to measure time that the valve is closed

**Week of 6/17** _week of conference_
EPA Grant Proposal:
- [ ] Edit and finish up grant proposal

Lab:
- [ ] Trace complete pressure cycles of manually opening and closing of the ram pump
- [ ] Operate the pump for an extended period of time (maybe over a weekend?)

Other:
- [ ] Work on research report

closed 2019-06-28 by ChingPangggg