AguaClara/ram_pump #66: Finding Terminal Velocity

cheertsang, 2019-03-28:
7.35s 15.1416471 L = 2.06008804 L/s
5.85s 10.2L = 1.74358974 L/s
5.99 s 9.9 L = 1.8 L/s
^^ disregard these first three trials

6.09 s 9L = 1.47783251 L/s
5.72 s 9L = 1.57342657 L/s
5.63 s 8.6L =1.52753108 L/s
6.11 s 9.1L = 1.4893617 L/s
5.91 s 9L = 1.52284264 L/s

Average of the last five trials = **1.518199 L/s**

cheertsang, 2019-04-08:
flow_rate = 1.518199*u.L/u.s
area = pc.area_circle(2.372*u.cm)
velocity = (flow_rate/area).to(u.m/u.s)

3.4356542382769817 meter/second

cheertsang, 2019-04-08:
@monroews We were trying to measure the terminal velocity of the water in the drive pipe experimentally by measuring the flow rate and dividing by area of the drive pipe, but we were doing this without the ram pump and plate attached to the drive pipe. @mailemccann brought up the point that we should be measuring it with the ram pump and plate attached, to account for the headloss caused by the plate. **Does it make more sense to measure terminal velocity with the ram pump and the plate attached?**

monroews, 2019-04-08:
You need to know the velocity of the water in the drive pipe when you slam the waste valve shut. Therefore you need to measure the velocity of the water with the ram pump attached and the valve in the same exact position as you will be using when you operate the ram pump.

cheertsang, 2019-04-08:
250 mL/ 0.37 s = 0.675675676  L/s
630 mL/0.68 s = 0.926470588 L/s
620 mL/ 0.68 s = 0.911764706 L/s
480 mL/ 0.52 s = 0.923076923 L/s
560 mL/ 0.66 s = 0.848484848 L/s

monroews, 2019-04-08:
The biggest error here is the time. Can you measuring larger volumes so you can have a sample time of a couple of seconds at least?

cheertsang, 2019-04-08:
4.055 L/ 3.98 s
4.670 L/ 4.68 s
3.880 L/ 3.93 s
3.565 L/ 3.60 s
3.790 L/ 3.83 s

closed 2019-04-25 by cheertsang