AguaClara/ram_pump #67: Theoretical Volume of Water Pumped

cheertsang, 2019-04-11:
@monroews We are writing code to calculate the theoretical volume of water pumped during each cycle so that we can compare this to the experimental values. This is the equation that we're using, and we're using the terminal velocity we found by experimentation:

[image]

Our question is, **what is the delta t that we're using? Is it the time difference between opening and closing the valve?**
[image]

**Or is it the time difference when water is actually being pumped into the air chamber?**
(between roughly 0.02 and 0.07 s) 
[image]

Thank you!

monroews, 2019-04-11:
The theoretical maximum value that you can pump is based on the time required to decelerate the water that is traveling at the terminal velocity given that you are applying a pressure to the bottom of the drive pipe. 

Use the difference in pressure between the static pressure in the drive pipe and the pressure in the high pressure system to find the pressure that is decelerating the flow in the drive pipe. In an ideal system it would decelerate at a constant rate because the pressure would be constant.  I believe you derived the equation to calculate this delta t in our previous meeting.

monroews, 2019-04-11:
The empirical measurement of the delta t is close to 0.04 seconds as shown in the second graph.

cheertsang, 2019-04-12:
We derived this expression for delta t:
[image]

which we then plugged into the volume equation for the volume of water pumped per cycle:
[image]
[image]
where h is the height of the drive pipe, and delta h is the difference in height between the height we want to pump water and the water level of the head tank (initial head in air chamber - water level in head tank = 600 m - 2.06 m)

monroews, 2019-04-12:
The equations look great. How long is the calculated delta t compared with the measured delta t? Now you can calculate efficiency of pumping too!

cheertsang, 2019-04-12:
**The theoretical time period in which water is being pumped into the air chamber is 0.0976 second**
**The theoretical amount of water pumped is: 48.73 milliliter**
We are skeptical because the time period and the amount of water pumped were a lot greater than the experimental values we got. In our experiments, the volume of water pumped per 1 cycle is 0.6337 milliliter. 

```
v_f = 2.26*u.m/u.s #terminal velocity, experimentally determined
area = pc.area_circle(.02372*u.m) #area of drive pipe

#height of the drive pipe
h = 1.74*u.m

#height difference between height we want to pump water and water level of head tank
h1 = 2.06*u.m #height from bottom of drive pipe to top of water level in head tank
h2 = airchamber[0] #around 600 cm, make sure you run airchamber from first cell
delta_h = (h2 - h1).to(u.m)

delta_t = (v_f*h)/(c.GRAVITY*delta_h)
print('The theoretical time period in which water is being pumped into the air chamber is ' + str(delta_t))

volume = (0.5*area*v_f*delta_t).to(u.milliliter)

print('The theoretical amount of water pumped is: ' + str(volume))
```

monroews, 2019-04-15:
Now the challenge is to learn why the ram pump is sooooo inefficient. I bet you remember that we are doing this because we suspected that the ram pump was inefficient! 

I suspect that the pressure trace is still not capturing the very rapid pressure pulse fully. The water in the drive pipe is stopping MUCH faster (by about a factor of 3 - check this!) then what theory predicts. Why is the water stopping so much faster? We don't have any doubts about F=ma! So either the water isn't traveling as fast as you thought OR the pressure at the bottom of the drive pipe is much greater than you thought. The pressure trace DOES exceed 600 cm of water. I bet the pressure actually goes MUCH higher than 600 cm of water and that we are missing that very high pulse because our pressure sensor is to far from the drive pipe. 

The very high pressure required to move the water into the the high pressure side of the system is due to the head loss through the small check valve. So time to experiment with a bigger check valve! Do you have the plumbing supplies you need to build and test this?

closed 2019-07-31 by alyssaju