AguaClara/ram_pump #69: Meeting with Monroe (4/18)

ChingPangggg, 2019-04-20:
Grant report and proposal
- include report from this semester
- brainstorm for proposal

- more data resolution for the pump cycle
- make pressure sensor closer?
- Ask sensor dev for the new pressure sensors
- buy new sensors with higher failure
- pressure keeps increasing in the pressure sensor and water shot up
- the one in the ram pump broke as we closes the valve
- https://www.omega.com/en-us/search/?text=px26
- https://assets.omega.com/manuals/test-and-measurement-equipment/pressure/pressure-transducers/M1608.pdf

- there might be a spike that's not read by the pressure sensor and higher than 42m of water
- head loss going through valve straight down is equal to the one going side way
- making the check valve sideways as big as the valve going down (diameter) of drive pipe to reduce head loss
- do a head loss calculations (estimate k value for the small diameter)

- are we wasting the energy through head loss
- do we get a pressure that would break the sensor with this head loss

- To make the diameter of the effluent the same as the drive pipe-->
- t-shape design and get male threaded to match female threaded check valve
(Standard-Wall Chrome-Plated Brass Threaded Pipe Nipples and Pipe)

- Look into swing (flap) vs spring check valve
- spring might be better in closing the check valve when the flow slows down

- put that pressure is higher than pressure rating in our report

rod is fabricated by us

union between the two check valves
get two brass nipples and brass union

get the two forces (do we have them already)
min to open and max to close again (2 eqs 2 unknowns and assume delta x)
min force to open is also lower than the max force to close the valve
assume the length of the spring and find the k constant?
k is a property of the length of the spring

ChingPangggg, 2019-04-25:
@monroews Are we getting differential or gauge pressure sensor?

monroews, 2019-04-25:
Differential

closed 2019-07-31 by alyssaju