AguaClara/ram_pump #71: Fabrication of new ram pump

ChingPangggg, 2019-04-29:
@monroews We are not sure about what to buy for fabrication of a new ram pump.
1. First of all, for the rod that we needed to fabricate on our own, are we buying a fully threaded rod and then cut it down?
2. For the design that involve pipe union, should we get a threaded T-joint, and a brass or PVC pipe union ?

ChingPangggg, 2019-04-29:
@monroews And is this what we should order? https://www.mcmaster.com/9162k476

monroews, 2019-04-29:
COnsider meeting with Paul Charles to see if he would have a recommendation for an easier way to fabricate the rod.  We previously purchased a threaded rod and then cut it down.
The union looks good.
You can probably get a threaded Tee and short pipe nipples to connect between T and check valves.
You will also need nipples to connect union to check valves.

ChingPangggg, 2019-04-29:
Also, since we now want the effluent to be 1" pipe size, we will need to fabricate a new air chamber, and then use an elbow to connect the air chamber with the check valve. Is this correct?

monroews, 2019-04-30:
Draw a sketch and that will make it clear what pieces you need.

ChingPangggg, 2019-05-07:
Tapping hole for pressure sensor on T-joint:
Drill bit: 7/16
Tap: 1/4in- 18 (or 16?)

closed 2019-06-05 by ChingPangggg