AguaClara/ram_pump #73: EPA P3 Grant Proposal: Phase II Brainstorming

alyssaju, 2019-06-04:
###Budget Ideas:
- Jan Honduras Trip for two students per year for two years
- Need to test spring vs weight system in Honduran AC plant
- 4+ iterations of varying ram pump designs
updated pulley system
- using solely weights instead of springs and fine tune using weights _use cable with two weights (based on the forces for open and close states) with a set distance apart from one another_
  - Spring system mounted in the plant
  - Would reduce number of check valves and issues

ChingPangggg, 2019-06-05:
Additional Ideas:
- Include nice graphics in the grant
- We have approximately $46875 in total
- how to make it easier for fine adjustment of both the spring tension and forces

- pulley system vs spring force?
- using a weight and pulley system instead of a spring system
--> allow to tune without opening the ram pump because can tune the weights

- put spring up in the plant --> spring system mounted in the plant (stainless steel cable)
eliminate one of the check valves at the ram pump
- work with template to incorporate ram pump in the template to include in the grant proposal
-talk about running the next generation ram pump in the phase I portion
- come up with how much money it would take to buy the components and then double it to take to honduras

ChingPangggg, 2019-06-07:
- Buying more pressure sensors or trying out different pressure sensors
- trying out different variables
- Full scale model?
- building a system for the conventional ram pump design and one for the vertical ram pump to compare and see the problems of the current conventional ram pump
- measure how efficient is the current ram pump working in the plant
- sending someone to Honduras to specifically work on ram pump, and interview operators and talking with them about why they don't use it and figuring out ways to make it easier to use
- running tests on the ones that are used in the field
- measuring the efficiency in the field, then coming back to Cornell to replicate it
- Replicate the ones they use in Honduras in the lab at Cornell
- One team could focus on the social aspect of figuring out how to make the ram pump something that operators will actually use, and one could focus on running tests on the conventional ram pump and calculating it's current efficiency.

closed 2019-06-26 by ChingPangggg