AguaClara/ram_pump #76: Purchasing new springs

ChingPangggg, 2019-06-27:
Distance from bottom holder (check valve) for bottom face of union: 9.6 cm
Position bottom jam nut at 4cm from top face of union (when the plate valve is opened)
Minimum distance of spring: 9.65 cm - 1.2 cm - 4 cm = 4.45 cm (1.73228 in)
Min spring constant: 6.924 N/cm (0.6128 lbs/in)

ChingPangggg, 2019-06-28:
[McMaster Spring filtered catelog.xlsx](https://github.com/AguaClara/ram_pump/files/3340048/McMaster.Spring.filtered.catelog.xlsx)
Ordered 4 springs (Jun28 19) with one of them only work of one of the spring constant (6.28)

ChingPangggg, 2019-07-02:
If plate valve remains opened--> spring force is too strong--> k' is too big--> move jam nuts up
If plate valve remains closed--> spring force is too weak--> k' is too small--> move jam nut down

alyssaju, 2019-07-02:
[Test 1](https://github.com/AguaClara/ram_pump/blob/master/Summer%202019/Data%20Collection/7-2-2019_spring_K146_test1.xls): 
- effluent valve was closed; initially drive pipe was closed and then opened
- allowed water to run through pump but water leaked through joints (where ram pump connects to the drive pipe & nipple joints in ram pump) 

[Test 2](https://github.com/AguaClara/ram_pump/blob/master/Summer%202019/Data%20Collection/7-2-2019_spring_K146_test2.xls): 
- effluent valve was closed
- tightened joints and eliminated most water leakage
- water filled into the air chamber at a faster rate; take a look at the spikes in the graphs

[Test 3](https://github.com/AguaClara/ram_pump/blob/master/Summer%202019/Data%20Collection/7-2-2019_spring_K146_test3.xls)

alyssaju, 2019-07-10:
Position bottom jam nut(including washer) at 4cm from top face of union (when plate valve is opened)
test1
test2

Position bottom jam nut(including washer) at 4.3cm from top face of union (when plate valve is opened)
test3

alyssaju, 2019-07-11:
[Test 1](https://github.com/AguaClara/ram_pump/blob/master/Summer%202019/Data%20Collection/7-11-2019_spring_K159_test1.xls)
- spring cut 14.95N
- positioned stopper at 4.55cm from top face of union (when plate valve is opened)
- water level in air chamber increased by 0.1cm

[Test 2](https://github.com/AguaClara/ram_pump/blob/master/Summer%202019/Data%20Collection/7-11-2019_spring_K159_test2.xls)
- positioned stopper at 4.00cm from top face of union
- water level in air chamber increased by 0.1cm 

[Test 3](https://github.com/AguaClara/ram_pump/blob/master/Summer%202019/Data%20Collection/7-11-2019_spring_K159_test3.xls)
- positioned stopper at 3.00cm from top face of union
- started at 6.0cm

alyssaju, 2019-07-12:
[Test 2](https://github.com/AguaClara/ram_pump/blob/master/Summer%202019/Data%20Collection/7-12-2019_spring_K428_test2.xls)
- positioned stopper at 2.2cm from top face of union
- initial height of water in air chamber (start): 18cm
- final height of water in air chamber: 84cm

ChingPangggg, 2019-08-02:
From rod end to bottom stopper: 8.45 mm +3.02 mm = 11.46 mm = 1.146 cm

closed 2021-03-11 by ChingPangggg