AguaClara/ram_pump #77: Things to Do (Summer)

ChingPangggg, 2019-06-28:
- [ ]  Redo intrinsic spring value calculations
- [ ]  Order springs that fit into the intrinsic k value graph
- [ ]  Redo head loss calculations
- [ ]  Trace complete pressure cycles of manually opening and closing of the ram pump
- [ ]  Operate the pump for an extended period of time (maybe over a weekend?)

alyssaju, 2019-07-15:
Graphing one cycle of acceleration of driving head:

- [ ] Find flow rate when plate closes
- [ ] Calculate terminal velocity using Q = v*A
- [ ] Calculate time that the plate was closed (when P_wastevalve > P_airchamber)
- [ ] Find slope of deceleration after plate closes (a = v_terminal/t)
- [ ] Modify acceleration input in the code by plugging in the correct v_terminal
- [ ] Graph and calculate the volume underneath the deceleration curve

alyssaju, 2019-07-17:
Update schematic in summer 2019 report to account for new air chamber design

alyssaju, 2019-07-26:
- [ ] Test with strong spring force
- [ ] Brainstorm better precision and measuring methods for finetuning the spring force with the lock nuts

alyssaju, 2019-08-02:
Increase P + decrease I in PID to make it respond faster

closed 2021-03-11 by ChingPangggg