AguaClara/ram_pump #79: Spring Force Adjustment

ChingPangggg, 2019-07-10:
We ran the ram pump today with setting the F_closed as 14.95 N. However, the spring force is still too weak as you can see from the graphs that the plate valve was closed the entire time. For the first graph, I set the jam nut 4cm from the face of union, and for the second one I adjusted the jam nut to the shortest distance I could, which is 4.3 cm from the face of union, but the spring force was still too weak. In addition, I think cutting the spring has caused the spring to buckle, which I am not sure how much would this affect the results. Do you have any suggestions on how to go about these two problems?
[image]
[image]
@monroews

alyssaju, 2019-07-11:
MONROE!! Exciting news. But to preface, we made a conversion error when calculating the correct length of the spring. We fixed our calculations and used the correct spring (9657K428). The pump cycled quite loudly and pumped a significant amount of water. Here is our pressure graph for this spring:

[image]

ChingPangggg, 2019-07-11:
For cycle shown above:
F_opened = 0 N since there is no initial compression (x_1 = 0cm)
F_closed = 12.236 N (x_2 = 1.37 cm)

monroews, 2019-07-12:
What was your expectation for the cycle time? It appears that the pump achieved 20 m of head! That is impressive. What role might F_opened have in the cycle time?

monroews, 2019-07-12:
Congratulations on getting the pump to cycle automatically! This is a big step forward!
One of the next steps is to write the code to calculate the flow rate as a function of the pump pressure.
Consider using meters as the unit on the y axis.
What is happening in the air chamber with each pump cycle? Is water flowing into and then back out of the air chamber or why is the pressure cycling? Is the data good enough to be able to calculate the amount pumped per cycle?

ChingPangggg, 2019-07-12:
`
url5 = 'https://raw.githubusercontent.com/AguaClara/ram_pump/master/Summer%202019/Data%20Collection/7-11-2019_spring_K428_test1.xls'

start = 29600 #should be more than 'start'
end = 29700 #should be less than 'stop'

x = (pp.column_of_time(url5,start,end)).to(u.s)

pressure = pp.column_of_data(url5, start, 1, end, 'cm')
airchamber = pp.column_of_data(url5, start, 2, end, 'cm')

plt.plot(x, pressure, '-')
plt.plot(x, airchamber, '-')

idx = (np.argwhere(np.diff(np.sign(pressure - airchamber)))).flatten()
plt.plot(x[idx], pressure[idx], 'o')

idx2 = (np.argwhere(np.diff(np.sign(pressure - airchamber)))+1).flatten()
plt.plot(x[idx2], pressure[idx2], 'o')
plt.show()
print((x[idx[0]]+x[idx2[0]])/2)
print((x[idx[1]]+x[idx2[1]])/2)
`
[image]

Using above code, the first intersection point was found to be 0.087 second and the second was 0.093 second so the pumping cycle was 0.006 second for this specific cycle. But similar code can be applied for every 0.4 or 0.5 until next cycle is found?

closed 2019-08-09 by alyssaju