AguaClara/ram_pump #81: Running Pump Effluent with PID

ChingPangggg, 2019-07-25:
hex nut located at 2.15 cm from union face when plate valve is opened and 3.35 cm from union face when plate valve is closed
Measured effluent flow using new design and spring K428:
913 mL /19.39 seconds = 47.0861269 mL/s = 0.471 L/s

[First data collection](https://raw.githubusercontent.com/AguaClara/ram_pump/master/Summer%202019/Data%20Collection/New%20Air%20Chamber%20Design/7-25-2019_spring_K428_test1.xls): (no PID) 
start - end: 610 mL

[Second data collection](https://raw.githubusercontent.com/AguaClara/ram_pump/master/Summer%202019/Data%20Collection/New%20Air%20Chamber%20Design/7-25-2019_spring_K428_test2.xls): (no PID)
3 in pipe: 50.3 cm (start) (pass high pressure effluent) - 54.5 cm (end) (high pressure effluent valve closed)

alyssaju, 2019-07-26:
[Fourth data collection](https://raw.githubusercontent.com/AguaClara/ram_pump/master/Summer%202019/Data%20Collection/New%20Air%20Chamber%20Design/7-26-2019_spring_K428_test1.xls): 
- two pumps attached to air chamber
- distance of exposed rod from coupling nut = 99.94mm (1.22mm is about half a turn of coupling nut)

[Fifth data collection](https://raw.githubusercontent.com/AguaClara/ram_pump/master/Summer%202019/Data%20Collection/New%20Air%20Chamber%20Design/7-26-2019_spring_K428_test2.xls):
- distance of exposed rod from coupling nut = 98.55mm

[Sixth data collection](https://raw.githubusercontent.com/AguaClara/ram_pump/master/Summer%202019/Data%20Collection/New%20Air%20Chamber%20Design/7-26-2019_spring_K428_test3.xls):
- distance of exposed rod from coupling nut = 97.57mm

ChingPangggg, 2019-07-31:
**7-29-2019 spring K127**

First data collection:
- distance of exposed rod from coupling nut = 82.57mm
- started with empty air chamber

ChingPangggg, 2019-07-31:
**7-31-2019 spring K127**

[First data collection:](https://raw.githubusercontent.com/AguaClara/ram_pump/master/Summer%202019/Data%20Collection/New%20Air%20Chamber%20Design/7-31-2019_spring_K127_test3.xls)
- distance of exposed rod from end of rod to coupling nut = 100.12mm (too weak)
- 98.94mm (more consistent results)

ChingPangggg, 2019-08-05:
**8-05-2019 spring K127**

trial 1:
- empty chamber
- not pressurized with peristaltic pump

trial 2: 
- pressurized with peristaltic pump when water level was in the middle of T-joint
- water poured into the bottom air chamber as PID control was on; inconsistent pump-cycles but water level was maintained steadily below the T-joint

trial 3: 
- pressurized with peristaltic pump when water level was above the T-joint
- PID control was not responding as fast as the pump (observe cyclic linear decrease and increase of air chamber pressure in graph)

trial 4: 
- changed the PID control: higher P (increases the responsiveness) and lower I (increases the importance of historical data which stabilizes our fluctuating data) --> P = 1.2; I = 100m
- pressurized with peristaltic pump when water level was above the T-joint

trial 5:
- rerun of trial 4 
- cycle was pretty consistent, but it seemed inconsistent due to air bubbles that would form as air entered through the 1/4in check valve (test this hypothesis)

closed 2021-03-11 by ChingPangggg