AguaClara/ram_pump #83: Inconsistent Pump Cycles - Spring K127

ChingPangggg, 2019-07-31:
[7-31-19_spring_K127_trial 3](https://raw.githubusercontent.com/AguaClara/ram_pump/master/Summer%202019/Data%20Collection/New%20Air%20Chamber%20Design/7-31-2019_spring_K127_test3.xls)

Beginning when effluent valve was shut & air chamber was being pressurized w/ peristaltic pump:
[image]

After effluent valve was open & PID control was used to maintain pressure & air chamber was not being manually pressurized by pump:
(Zoom 1)
[image]

(Zoom 2)
[image]

We noticed that there were some inconsistent cycles (as shown in zoom 1 & 2). Some of the pressure spikes were too close together. We can't get a sense of what might be causing these weird patterns such as the second cycle in the graph shown directly above. The adjustment of the spring length seems to be fine since the cycles were consistent in the beginning of this data collection when the effluent valve was closed. 

@monroews

monroews, 2019-07-31:
Aren't these cycles much faster than optimal? Don't you want closer to a second for the cycle time?

alyssaju, 2019-07-31:
Yes, we'll work on increasing the cycle times, but we wanted to primarily focus on keeping the cycles consistent. Wouldn't changing the cycle time not affect the inconsistent cycles? What would cause the inconsistent time cycles if we kept all the initial conditions the same?

monroews, 2019-07-31:
Not sure, but I'd argue that there is no point in figuring this out UNLESS it is a problem when the cycle times are correct.

closed 2021-03-11 by ChingPangggg