AguaClara/ram_pump #89: Flow Rate Calculations for Sump Pump

AlyciaStorch, 2019-09-24:
[image]
Our pump is R5S-1 (last row)
For 6 feet = 1.83 m:
58 gallons per min = 3.66 liters/second

@monroews This is the only table I could find for our sump pump. This info is for 1-1/2" pipes, there is no info on different pipe diameters and I wasn't quite sure how to do a conversion). We got a theoretical flow rate of 1.587 liters/second through the drive pipe without the ram pump attached with our 2 m of head. We figured the ram pump only slows things down so if the sump pump could accommodate the flow rate without the ram pump, we would be fine with the ram pump attached. Do you think that the sump pump being able to pump twice as much as the amount that leaves the drive pipe is sufficient info to assume attaching a 1" pipe to the sump pump would be fine? Or do you have any tips for doing this conversion?

monroews, 2019-09-25:
The head loss in the 1 inch pipe at the ram pump terminal velocity is less than 3 m (or whatever the drive pipe height is). That will be true when the 1" pipe is installed on the centrifugal pump too! Thus the total head that the pump will have to provide is 3 m of height plus 3 m of head loss. That is 6 m = 18 feet.  THe corresponding flow rate for the pump is 30 gallons per minute. Does that make sense? The trick is to think of this in terms of head and head loss. You could, of course, calculate the head loss in the 1" pipe using the aguaclara code! That would be an even better approach!

AlyciaStorch, 2019-09-25:
@monroews Are you saying that instead of saying the sump pump has to pump to a head of 2 m AND the drive pipe is taking a flow rate corresponding to 2 m of head, you can just say the sump pump needs to pump to a head of 4 m? We are a little confused because we thought comparing the flow rate the pump can provide to the flow rate exiting through the drive pipe would clearly show us if the pump could accommodate our system or not. If we just look at flow rate for 4m of head are we just looking to see that the pump can in fact pump to that head? Also we are still a little confused as to where the diameter comes into play. Does the sump pump just pump a certain flow rate no matter what and changing the pipe diameter just changes the speed of the water? Wouldn't reducing the pipe diameter add head loss to the system and effectively increase the head we want the pump to pump to? Do we just need to account for that extra head loss or is there more to going from 1-1/2" to 1" diameter?

monroews, 2019-09-25:
Okay. I tried to explain that the head loss in the pipe that is connected to the sump pump was going to be less than 2 m and that is why it has to effectively add 4 meters of head to the water. You can EASILY calculate the head loss in the 1" pipe. I think you should do that so you see how this works. You should include both entrance (contraction from 1.5 to 1) and exit losses.

Once you calculate the head loss in 1" pipe at the flow that you need, then you add that headloss to the elevation head and then look up the total head in the table to see if the pump can deliver at least as much flow as you need.

The sump pump flow rate is a function of the head. It is not a constant displacement pump (like the peristaltic pumps).

closed 2021-03-11 by ChingPangggg