AguaClara/ram_pump #91: Lab Bench set-up for Ram Pump

ChingPangggg, 2019-10-10:
[image]

We talked to Paul about fitting ram pump experimental setup into out lab bench, and Paul advise us to not drill holes on the bench, and instead take the table off or find a smaller top that would only occupy 1/2 of the current table space so we can put the ram pump set up in the other half. Do you have any suggestion about this?

@monroews

monroews, 2019-10-10:
Why did he make this suggestion? Is it because drilling holes is hard or because he didn't want us to damage the lab benches?

Did he have any recommendations for an alternate top that we could drill holes through?

ChingPangggg, 2019-10-10:
> Why did he make this suggestion? Is it because drilling holes is hard or because he didn't want us to damage the lab benches?
> 
> Did he have any recommendations for an alternate top that we could drill holes through?

He said it is very difficult to drill through the current lab benches and he also said that it would probably require a lot of effort and a special hole saw. He have not recommended another table top that we can drill through, but he said we could put a wooden top on the other half of the table just so we still have part of the lab bench to use, so I think it would also be possible to drill through the wooden top if we want to.

closed 2021-03-11 by ChingPangggg